JTechPhotonics/J-Tech-Photonics-Laser-Tool #1: Error when trying to apply

AdNine87, 2020-06-11:
Python path configuration:
  PYTHONHOME = '/tmp/.mount_InkscaCWB7VM/usr/'
  PYTHONPATH = '/home/user/.config/inkscape/extensions:/tmp/.mount_InkscaCWB7VM/usr/lib/share/../../share/inkscape/extensions:/tmp/.mount_InkscaCWB7VM/usr/lib/share/../../share/inkscape/extensions/inkex/deprecated-simple'
  program name = '/usr/bin/python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/tmp/.mount_InkscaCWB7VM/usr'
  sys.base_exec_prefix = '/tmp/.mount_InkscaCWB7VM/usr'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/tmp/.mount_InkscaCWB7VM/usr'
  sys.exec_prefix = '/tmp/.mount_InkscaCWB7VM/usr'
  sys.path = [
    '/home/user/.config/inkscape/extensions',
    '/tmp/.mount_InkscaCWB7VM/usr/lib/share/../../share/inkscape/extensions',
    '/tmp/.mount_InkscaCWB7VM/usr/lib/share/../../share/inkscape/extensions/inkex/deprecated-simple',
    '/tmp/.mount_InkscaCWB7VM/usr/lib/python38.zip',
    '/tmp/.mount_InkscaCWB7VM/usr/lib/python3.8',
    '/tmp/.mount_InkscaCWB7VM/usr/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fd6fe424740 (most recent call first):
<no Python frame>
I'm on linux debian

PadLex, 2020-06-12:
Take a look at [this discussion](https://discuss.python.org/t/fatal-python-error-init-fs-encoding-failed-to-get-the-python-codec-of-the-filesystem-encoding/3173). It seems like your python path is miss configured.

PadLex, 2020-06-12:
Another [useful resource](https://www.google.com/url?sa=t&source=web&rct=j&url=https://stackoverflow.com/questions/5694706/py-initialize-fails-unable-to-load-the-file-system-codec&ved=2ahUKEwj5wLrhgfvpAhXDcJoKHVAoD_8QjjgwAnoECAIQAQ&usg=AOvVaw3YWjYQCupQ6T3WxtsKM-Fl&cshid=1591921742980). Regardless, it doesn't seem like it has anything to do with the extension. So if you don't object I'll close the issue.

closed 2020-06-19 by PadLex