JTechPhotonics/J-Tech-Photonics-Laser-Tool #2: Problems with Inkscape 1.0

bianchifan, 2020-06-15:
Scenario 1: Windows 10 x64, Inkscape 1.0 portable freshly unzipped
SVG: Multilayer SVG, all layers unvisible but TWO! cutting layers

Result:
Several deprecation messages and a weird screen ouput in preview, the plug took use of some hided layers and created its own positions, 
[image]
scalings and proportions.

A little closer view with the shades of the two cutting layers "cut1" and "Cut Seg Lockout" within the DIN A4 frame
[image]
[image]

The generated gcode file includes 300kB Data.

Next try, same svg with all unvisible layers deleted:
[image]
[image]

Odd placement but correct scalings and dimensions, the generated gcode counts 8,6 kB ;)

Scenario 2 : Windows 7x32 within Virtualbox on Ubuntu Host(20.04), Inkscape 1.0 portable freshly unzipped
SVG: One layer with an outline for cutting, burning..
same screen results as with scenario 3

Scenario 3: Ubuntu 20.04 
deprecation warnings, wrong scaling, wrong placing
[image]
[image]

While win7 gcode (scenario 3) eats 2.9 kB Ubuntu's gcode is a little smaller, 2.7 kB
Both files match in line numbers but differ in "G1" code lines..
[image]

To compare:
With Inkscape 0.92.5/Ubuntu Studio 19.10 the screen output is correct (within borders, lines&arrows maybe a little big..) and the gcode eats only 2.5 K while tallying same lines but differing of course..
[image]

[image]

jithivk, 2020-07-02:
raceback (most recent call last):
  File "laser.py", line 78, in <module>
    from inkex.paths import CubicSuperPath  # noqa
ImportError: No module named paths
 this is the error what i have

mountwebs, 2020-07-11:
I get some similar results to OPs. The extension mostly works fine, but when I tried to use it after applying Hershey text, I got the same script error as @bianchifan and the result in inkscape was a lot of huge displaced paths.

macOS 10.13, Inkscape 1.0.

PadLex, 2020-07-13:
> raceback (most recent call last):
> File "laser.py", line 78, in
> from inkex.paths import CubicSuperPath # noqa
> ImportError: No module named paths
> this is the error what i have

Jithivk, check out this issue. [ImportError: No module named paths #3](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/issues/3)

PadLex, 2020-07-13:
@bianchifan This is really weird. I'm not even sure how to start reproducing this bug. The output really shouldn't vary between operating systems. Are you sure you kept the same Inkscape version, extension version, and settings between experiments?

PadLex, 2020-07-13:
In the meantime, could you share the SVG's which don't work? I can't work on this now but I'll try and reproduce your issue when I have a little time.

bianchifan, 2020-07-14:
> 
 The output really shouldn't vary between operating systems. Are you sure you kept the same Inkscape version, extension version, and settings between experiments?

Ubuntu 20.02:  Official Inkscape 1.0

Windows 10 x64
Windowx 7 x32 (Modern IE) / Virtualbox :  Inscape 1.0 poertable from PortableApps
[image]
[image]

Settings should be equal

bianchifan, 2020-07-14:
> 
> 
> In the meantime, could you share the SVG's which don't work?

In the meantime the svgs had changed due to to varied placing and new objects.
I can share the actual version:

[Fox32_decals_Set_portrait.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/4917757/Fox32_decals_Set_portrait.zip)
[image]

[Fox32_decals_Set_portrait_cut2.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/4917766/Fox32_decals_Set_portrait_cut2.zip)
[image]

JuPrgn, 2020-07-14:
Hi,
I have same warning with fresh Inkscape 1.0 install and corresponding pluggin 1.0 beta (win7 64).
In my case the preview is very far from the original drawing : 
[image]

I had already reported this bug with this file to JTP with previous 0.92 extension.

The source file is a part from another complex SVG and I only extracted the part that make this bug happen : 
[drawing_initial.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/4920053/drawing_initial.zip)
[image]

Please let me know if you need some more details.

bianchifan, 2020-07-14:
> @bianchifan It would be nice if you could give it a try and share if you have similar issue with my SVG

1st.. Ubuntu 20.04 / Inkscape 1.0

does not run due to syntax error
[image]

2nd.. Win7 x32 / Inkscape 1.0 portable version (PortableApps) via Virtualbox
[image]

PabloSica, 2020-07-15:
Hello friends. I use your plugin for laser engraving in InkScape 1.0 on a Catalina MacOs. 
In this version of MacOs I cannot install 0.92, only 1.0 For some reason after updating the Operating System I can no longer use the plugin. When I generate the file for recording, I receive the following message: 
File "laser.py", line 51
    def bezierslopeatt ((((bx0, by0), (bx1, by1), (bx2, by2), (bx3, by3)), t):
                       ^
SyntaxError: invalid syntax

I appreciate if you can help me.
Congratulations on the great job.
Thank you.
Pablo Sica

PadLex, 2020-07-27:
All of the above issues should be fixed in the latest release v1.1 patch. @bianchifan and everyone else, let me know if the problem has been resolved so I can close this issue.

bianchifan, 2020-07-28:
quick test of v1.1 patch...
- actions within Ubuntu 20.04 and Win7-Virtualbox(Modern IE) are equal, scaling & positioning seems to be ok :)
- still no difference between hidden and visible layers - all layers are routed
- the preview shines very ,very big, still - not able to drive in detailed view
[image]
[image]

PS:
only this should be routed
[image]

PadLex, 2020-08-07:
Perfect, the scaling issues were the main problem here. The fact that hidden layers aren't ignored is the subject of a different issue so I'll close this one.

closed 2020-08-07 by PadLex