JTechPhotonics/J-Tech-Photonics-Laser-Tool #4: Problem with path generated and warnings

JuPrgn, 2020-07-14:
Hi,
I have same warning than #2  with Inkscape 1.0 fresh install and corresponding pluggin 1.0 beta (win7 64).
[image]

In my case the preview is very far from the original drawing : 
[image]

I had already reported this bug with this file to JTP with previous 0.92 extension.

The source file is a part from another complex SVG and I only extracted the part that make this bug happen : 
[drawing_initial.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/4920053/drawing_initial.zip)
[image]

Please let me know if you need some more details.

PadLex, 2020-07-26:
Thanks for reporting this bug. I'm prioritizing resolving new bugs (1.0 only) because I'm better equipped to fix them. In the next release we'll try to improve the script's structure and readability so that we can try and tackle older bugs like this one.

PadLex, 2020-07-26:
Didn't mean to close this issue. I think it was automatically closed because the deprecation warnings should have been fixed in the master branch (I'll add a new minor release after some testing).

PadLex, 2020-07-27:
Nevermind, all of thesee issues should have been fixed with the drewler's commit. You can now download the latest 1.1 patch.

closed 2020-07-27 by PadLex

JuPrgn, 2020-08-06:
Thank you @drewler and @PadLex.