JTechPhotonics/J-Tech-Photonics-Laser-Tool #6: Does not ignore locked layer

horga83, 2020-07-16:
Version: BETA plugin for Inkscape 1.0

The old plugin ignored any locked layer which is correct.
This version includes the locked layer.

A locked layer is helpful as a layout grid for where the object should be processed with the laser but should not be included in the gcode created.

Thanks.

PadLex, 2020-07-26:
Hi, thank you for the bug report. Do you know what python version you're running?

horga83, 2020-07-26:
On Sun, 2020-07-26 at 09:19 -0700, Alex Padula wrote:
> Hi, thank you for the bug report. Do you know what python version
> you're running?
> 
> 
> 
My version of Python is 3.8

PadLex, 2020-07-26:
That's odd, I'll do some testing and try to fix this by the next minor release. In the meantime, if you need this feature urgently, you could try and use Python 2.

horga83, 2020-07-26:
Sorry I was on the run when I emailed and didn't have time to further
comment.
I am running Arch with latest updates.  It seems that the command to
run Inkscape is failing to produce a file, if I run the inkscape
command on the command line it works.  If I then plug that file into
your script it produces a png where all the white background turns out
black.  The laser just starts burning a fill page of black.
That is as far as I've got, haven't had time to spend on it again,
wedding in the family sort of thing:-)
CheersGeorge

PadLex, 2020-07-27:
> Sorry I was on the run when I emailed and didn't have time to further comment. 

No worries at all, if anything thank you for responding abut 10x faster then I do ;)

> I am running Arch with latest updates. It seems that the command to run Inkscape is failing to produce a file, if I run the inkscape command on the command line it works. If I then plug that file [...]

Okay, now I'm a little more confused. So, to recap:

1. The first bug is that the plugin is supposed to ignore locked layers but does not. I've only updated the script to be compatible with Python 3 and Inkscape 1.0, so I don't know much about what is and is not an intended feature. It looks to me like that might have been an "unintended feature" (aka a bug) and locked layers should not have been ignored, what do you reckon? Regardless that seems like a good feature, we could make it toggle-able. Also, if you need a static grid you can create one from the Inkscape document settings. That's what I do, I think it's good practice to keep reference grids separate from the svg content.

2. The second bug is that the generated gcode is... filling the page with black? I don't understand the conditions for this to occur. What is the file that's failing to be produced? I apologize if I'm missing something obvious here but, since when does this script produce png's? Or anything other then gcode files? As far as I understand this process, laser.py parses an svg for path elements and converts them to a series of gcode commands using the settings from laser.inx. Are there more advanced use-cases which I am not aware of?

horga83, 2020-07-27:
Oh my goodness my sincere apologies, I have confused two bug reports.
 Please disregard my last email, it was supposed to be directed toward
305 engineering raster plugin.  You are correct.  The report we are
discussing is the locked layer one.
Need to stop being in a rush and focus a bit more:-)
CheersGeorge

PadLex, 2020-08-07:
No worries, I'll see if I can re-introduce this feature in the next update. If you can, feel free to give it a try yourself and shoot me a pull request.

closed 2021-04-07 by PadLex