JTechPhotonics/J-Tech-Photonics-Laser-Tool #10: JTech Inkscape Beta Extension for Inkscape 1.0 doesn't work!

oliverhbailey, 2020-07-28:
I have attempted to install this extension under Windows 10 Home Build 18362_19h1_release.190318-1202 to no avail. The documentation is in conflict. On the web page it states the contents of the .zip file should be extracted to the System Extensions folder under Edit->Preferences->User->Extensions path which is Users/Username/AppData/Roaming/inkscape in inkscape version 1.0 (4035a4fb49, 2020-05-01).

ironically, that folder does not exist prior to making an attempt to install the JTech extension. There are other extensions already installed but they are either installed by an installer program or have been installed at the time inkscape was installed.

My questions are as follows:

1) Is this still a viable extension
2) Has it been maintained across inkscape versions (There is little comment after 2018)
3) Does there plan to be ongoing support for this plugin now and going forward.
4) Has ANYONE else reported this. The current version of inkscape was released almost eight months ago.

Since there seems to be a recent comment about moving to python 3, this may not be a very widely used plugin.

Please advise or fix the issue.

Thank You.

PadLex, 2020-08-04:
As far as I know, the original extension (0.9) hadn't been updated in a while. This repository was created recently to update and maintain the extension for future Inkscape versions. So yes, the extension is and will be actively maintained.

JuPrgn, 2020-08-06:
On my laptop using Inkscape 1.0 the extension folder is on c:/program files/Inkscape/share/Inkscape/extensions and i put the files from the zip inside this folder.

oliverhbailey, 2020-08-06:
I uninstalled version 1.0. I know the release notes indicate the migration
to python 3 is progressing through versions 0.92 and 1.0. I did install
another extension for laser cutting from another vendor yesterday, and it
worked just fine.

Oliver Bailey

oliverhbailey, 2020-08-06:
Juprgn

I checked the path for a second time on the current 0.92 extension path. It is as follows on Windows 10

C:\Program Files\Inkscape\share\extensions

This path makes sense. There is no need for a second inkscape folder within the inkscape application folder since it is already within the proper application. However, that may be why the instructions state to use the path defined by program preferences->system/path/extensions.

closed 2020-08-06 by oliverhbailey

JuPrgn, 2020-08-06:
The path I gave you is from Inkscape 1.0 Preference dialog. Nice you find the correct one.

oliverhbailey, 2020-08-06:
I already had the correct path for 0.92.

oliverhbailey, 2020-08-06:
Hi Alex,
I found a wiki that the inkscape developers put up for those having
problems with extensions executing between 0.92 and 1.00. I read through it
and am putting the link below.

https://wiki.inkscape.org/wiki/index.php?title=Updating_your_Extension_for_1.0

Regards,
Oliver Bailey

JayC2469, 2023-01-31:
Hi all! I thought it best to respond to this thread than create a new one, as my issue is applicable to this one (I think!)
I am having  issues getting **Anything** to happen when selecting either the Jtech Photonics and/or the Endurance Lasers extensions to generate Gcode for Laser Etching or cutting. The extensions all appear in the list but the current latest version of Inkscape I am using (1.2.2), is not playing ball, with regards to the Gcode Generating Extensions mentioned.
Any help would be  **Immensely**  appreciated