JTechPhotonics/J-Tech-Photonics-Laser-Tool #11: Inkscape 0.92 and JTech Laser driver 1.0 with 1.1 update doesn't work.

oliverhbailey, 2020-07-29:
After the problems with inkscape 1.0 and the JTech laser driver, I uninstalled version 1.0 and re-installed 0.92. However, when executing the tutorial as shown, an error is generated when creating the Text that no Transform function exists. Is there an additional python library that needs to be installed??

And a prior version of this extension did work at one time, over a year ago.

PadLex, 2020-08-04:
Hi oliver. Could you copy-paste the error you received? Is it identical both in 0.92 and 1.0? Are you using Inkscape 0.9 or 1.0?

oliverhbailey, 2020-08-04:
Hello Alex,
Actually I have tried both 0.92 and 1.0. They both generate eros that have
some similarities and some differences. I will have to recreate the errors
and email you later today. I read the release notes on the latest 0.92 and
1.0 inkscape. They are moving away from python 2.7 and changing the folder
structures, which I suspect is the root of the problem. The latest 0.92
version started the transition and they finished it with 1.0. They have
added a download button for plugins which has quite a few items in the
list, the Jtech drivers are there. It would appear this is a problem with
the drivers being maintained to current levels. I will be happy to email
the current errors that I got later today, I am simply working around the
problems. I was going to fix the drivers, but after 45 years of hardware
and software development, I decided to stand back and find an alternate
solution. This has come to surface with lasers and 3D printing systems, and
decided to take a different approach. If you update the drivers and want
someone to test, I would be happy to take the quick test route.

You will have the errors by days end.

Regards,
Oliver Bailey

oliverhbailey, 2020-08-04:
Hello Alex,
There are different problems using the jtech inkscape extension depending
on which operating system you're using. The point of confusion is where the
extensions get installed. On the jtech page the following is the
installation instructionsL

T*his plug-in will convert a path object in inkscape into a G Code file for
use in your laser upgraded machine.  Put the contents of this .zip folder
into the “inkscape\share\extensions” folder.*

On Linux there is no such folder. The path on Linux is as follows:

/usr/share/applications/inkscape/extensions

If you extract the files using the path, it creates a Jtech folder and the
.inx and .py files have to be copied to the "extensions" folder above.
Then start inkscape and create the first test text program in the tutorials
and the attached error box will pop up:

[image: image.png]

this is not the same error that is displayed in Windows 10. I will
forward that error tomorrow morning.
This was a clean install of inkscape on Debian 10, using apt to install the
current repository version.

I hope this helps to get started.

regards,
Oliver Bailey

oliverhbailey, 2020-08-04:
Hi Alex,
I completely removed version 0.92 and any remnant folders and reinstalled
inkscape 0.92 followed by the jtech extension. After running inkscape 0.93
and running the jtech extension, the error message generates is below:

Traceback (most recent call last):
  File "laser.py", line 78, in <module>
    from inkex.paths import CubicSuperPath  # noqa
ImportError: No module named paths

This is the WIndows 10 version.

Sincerely,
Oliver Bailey

PadLex, 2020-08-07:
Hi Oliver. I recommend following the instructions laid out in this repository's README.md file. The error you received usually occurs when you install a recent version of the extension with an old version of Inkscape. I recommend Installing the latest version of Inkscape (1.x) and the latest version of this extension (currently 1.1).

PadLex, 2020-08-07:
Just to make sure we're on the same page, this is an open-source extension for Inkscape. It was originally developed by Jtech but is now maintained by the community. I'm not sure what Jtech drivers refers to.

oliverhbailey, 2020-08-07:
Hi Alex,
I did what you recommended and removed version 0.92.5, and installed
version 1.0. I also installed the extensions into the proper extension
folder this time. When I went through the first tutorial by jtech, the
following error was generated. I have no idea why that file cannot be
written to. The file does not exist in the directory either. I would guess
that this is a Windows 10 thing.

laser.py:1360: DeprecationWarning: Effect.unittouu is now a method in the
svg document. Use `self.svg.unittouu(str)` instead.
  doc_height = self.unittouu(self.document.getroot().xpath('@height',
namespaces=inkex.NSS)[0])
Can not write to specified file!
c:\output_0001.gcode

Regards,
Oliver Bailey

oliverhbailey, 2020-08-07:
Hello again Alex,
I finally got the extension to work.

First, If inkscape 0.92.5 is uninstalled the /program
files/inkscape/share/extension folder is not deleted. If inkscape 1.0 is
installed afterwards, the folder /program
files/inkscape/share/inkscape/extensions is created. To make the extension
work, add the files into the later folder, not the first folder, and run
the program. Then the tutorial will work correctly.

Two caveats. First, under Windows 10, the inkscape program has to be
executed as the administrator or the gcode file cannot be created. Second,
The other error still exists, its just no longer fatal.

Regards,
Oliver Bailey

closed 2021-04-07 by PadLex