JTechPhotonics/J-Tech-Photonics-Laser-Tool #12: Pass depth calculated incorrectly

tarmogr, 2020-08-04:
I made a gcode with 4 passes and 0,0mm pass depth. To my surprize my printer crashed into the material below on second pass.

When opening the file I found this line:
`G1 Z-2.7755598776686065e-17`

Of course my marlin based printer took it as -2.77mm and ignored the e-17 as would be expected.

PadLex, 2020-08-07:
I'm struggling to reproduce this bug. I tried setting the pass depth to 0.0 and got this every time:
`G1 Z-0`
What extension and Inkscape versions are you using?

tarmogr, 2020-08-07:
Plugin: J-Tech-Photonics-Laser-Tool-1.0-beta_ink0.9
Downloaded 21.07.2020
Inkscape: 1.0 (4035a4fb49, 2020-05-01)

I can reproduce it every time.

tarmogr, 2020-08-07:
Ok I did more investigating. I deleted the preferences.xml file and reopened Inkscape, all works as expected now. I copied the old preferences.xml back and problem was back. I also found this on line 976:
    ` jtechphotonics.com.pass-depth="2.77556e-17"`
The question is how did it get there? Even though I have 0,0 set in the field.
Iuploaded my preferences file (renamed to .txt)
[preferences.txt](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/5042763/preferences.txt)

tarmogr, 2020-08-07:
Also another test. I deleted preferences.xml in C:\Users...AppData\Roaming\inkscape. Opened Inkscape and generated gcode with 1mm pass depth and 4 passes. Then decreased pass depth to 0,0. Result in gcode:
`G1 Z-1.3877787807814457e-16`

tarmogr, 2020-08-07:
I did following changes and this seems to cure the problem
Line ~565, Before:
`            gcode += "G91\nG1 Z-" + self.options.pass_depth + "\nG90\n" + gcode_pass`
Change to:
`            gcode += "G91\nG1 Z-{:.1f}\nG90\n{}".format(self.options.pass_depth, gcode_pass)`

Line ~627, Before:
`            {"name": "--pass-depth", "type": str, "dest": "pass_depth", "default": 1,
             "help": "Depth of laser cut"},`
Change to:
` {"name": "--pass-depth", "type": float, "dest": "pass_depth", "default": 1,
             "help": "Depth of laser cut"},`

I think the same has to be done with "power_delay" parameter. For example if I enter 0,1 the gcode produces:
`G4 P0.10000000149011612`

PadLex, 2020-08-18:
Thank you for figuring this out even though I'm so unresponsive this month. I'll add this change to the next patch.

closed 2021-04-07 by PadLex