JTechPhotonics/J-Tech-Photonics-Laser-Tool #17: Problem with inkscape 1.0

Dimdimtri, 2020-11-03:
Hello everyone, I just installed inkscape 1.0, if I draw a rectangle the plugin works fine. However if I import an external svg it gives me this error and the file is completely wrong. 

The same file on inkscape 0.92 works fine, I had to install inkscape 1.0 on my imac 10.15.

thank you in advance for your help.

laser.py:928: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  t = simpletransform.parseTransform(t)
laser.py:938: DeprecationWarning: simpletransform.applyTransformToPath -> Path(path).transform(mat)
  simpletransform.applyTransformToPath(trans, csp)
laser.py:295: DeprecationWarning: inkex.bezier.bezierlength -> Get length of bezier curve
  return bezmisc.bezierlength(bez, tolerance)

topogigio, 2020-12-26:
Hi,
I have same problem with Hershey font.
Thanks

Very big and strange tracks are generated after error:

laser.py:928: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  t = simpletransform.parseTransform(t)
laser.py:929: DeprecationWarning: simpletransform.composeTransform -> Transform(M1) * Transform(M2)
  trans = simpletransform.composeTransform(t, trans) if trans != [] else t
laser.py:938: DeprecationWarning: simpletransform.applyTransformToPath -> Path(path).transform(mat)
  simpletransform.applyTransformToPath(trans, csp)

aspeteRakete, 2020-12-31:
[image]
It seems that the center point of the object is translated to x=0 y=document_height

PadLex, 2021-04-06:
Should be fixed with the 2.0 release. Can you confirm?

closed 2021-04-07 by PadLex