JTechPhotonics/J-Tech-Photonics-Laser-Tool #21: Not working with Inkscape 1.0.1

goinheix, 2020-12-14:
I tried both ways, with a folder within the /share/extensions folder, and also dropping the files there. But it doesn't show up on my extensions menu. Am I doing something wrong, is there some tweak I'm missing, or it's just not working on 1.0.1 yet?

Thank you

TheGermanUSA, 2020-12-20:
Same issue here.

wren42, 2020-12-29:
I'm also having trouble with Inkscape 1.0.1.

I had no problem installing the extension in my home directory (~/.config/inkscape/extensions/). It shows up in the Inkscape extensions menu, but it still doesn't work right. Looks like the code's using some deprecated APIs, and that the extension's options aren't populated as the code expects.

Stack trace here:
laser.py:1360: DeprecationWarning: Effect.unittouu is now a method in the svg document. Use `self.svg.unittouu(str)` instead.
  doc_height = self.unittouu(self.document.getroot().xpath('@height', namespaces=inkex.NSS)[0])
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "laser.py", line 1459, in <module>
    e.run()
  File "/usr/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "laser.py", line 1452, in effect
    self.laser()
  File "laser.py", line 1300, in laser
    self.check_dir()
  File "laser.py", line 795, in check_dir
    if self.options.directory[-1] not in ["/", "\\"]:
IndexError: string index out of range

nineff, 2021-01-15:
also, no problem here with Inkscape 1.0.1. The files should be located in 
`C:\Program Files\Inkscape\share\inkscape\extensions` or similar on windows.

marciopmm, 2021-01-18:
Same issue here, or similar.

Got this error message:

Traceback (most recent call last):
  File "laser.py", line 1466, in <module>
    e.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "laser.py", line 1459, in effect
    self.laser()
  File "laser.py", line 1307, in laser
    self.check_dir()
  File "laser.py", line 802, in check_dir
    if self.options.directory[-1] not in ["/", "\\"]:
IndexError: string index out of range

wren42, 2021-01-19:
Marcio, I had the same issue you had and did some debugging to figure it out. That's what you get if you don't specify an output directory. If you fill in that box in the dialog, you won't get the error and it should work.

PadLex, 2021-04-06:
Should be fixed with the 2.0 release. Can you confirm?

closed 2021-04-06 by PadLex