JTechPhotonics/J-Tech-Photonics-Laser-Tool #26: Version 2.0 release candidate - I need your help for bug testing and feature requests

PadLex, 2021-02-02:
Hi all. I wanted to apologies for my inactivity lately and to make an exciting announcement. Version 2.0 is around the corner!

I wanted to get peoples feedback on the v2.0-beta. Have you encountered any bugs using it in your usual workflow? Are there any features you suggest? And for the coders among you, feel free to suggest ways to improve code readability. And remember, pull requests are always welcome.

nineff, 2021-02-03:
I just downloaded it and noticed many changes.
for me, the most important feature is now gone: the ability to set the laser on/off commands. 
I'm using this extension to generate instructions for plotting, not cutting, so custom commands for on/off were really handy.

What is the difference between the Standard G-Code and the Fan-Controlled G-Code?

PadLex, 2021-02-03:
The Fan-Controlled-Gcode interface is a standard interface with the only difference being that the laser is powered on using the Fan-power gcode command instead of Tool-power.

But you make a good point. It's not a very intuitive system and I might as well let people choose the command. It's pretty easy, I'll do that tonight.

Thanks for your feedback! :)

nineff, 2021-02-03:
thanks, sounds great.
I've also looked into the code, and the underlying library svg-to-gcode can accept a custom header and footer, but neither of the two default G-Code Compilers makes use of it.

I see the debugging code is still there, but I can't seem to access it from Inkscape. Is there an easy way?

JuPrgn, 2021-02-04:
I am not sure this is what you are asking but I added documentation about header and footer some weeks ago : https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/blob/master/README.md#custom-g-code-header-and-footer

nineff, 2021-02-04:
> 
> 
> I am not sure this is what you are asking but I added documentation about header and footer some weeks ago : https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/blob/master/README.md#custom-g-code-header-and-footer

I was just looking at the code and saw that the G-Code compilers get called without those parameters set. Since the Custom Header and Footer files were only documented for the old branch I assumed that this feature was gone.

PadLex, 2021-02-04:
To clarify, the custom header/footer files are missing from the wrapper branch. I'll re-introduce the feature asap. I was planning on doing that last night however a few major bugs got in the way.

If all goes well, tomorrow I'll publish a new release candidate wich isn't missing any features. I'll still need your help finding bugs and useful features before It's ready for a full release.

PadLex, 2021-02-05:
Ok. I published my changes under the v2.0-beta.2 tag (still in the v2.0 release candidate).

Custom header and footer support was re-introduced as well as debug tracers and reference points. + A bunch of bug fixes.
There are two know bugs, one involves edge cases with ellipses and the other is that transform attributes are ignored.

nineff, 2021-02-12:
Ok I had a quick look at it now. I'll need to modify it for my usecase as I have to use completely different commands for "laser on" and "laser off" while the current code only allows changes to the powerlevel. But it should not be too hard to implement.

uwefrank, 2021-02-12:
Attached a svg file link.
https://my.hidrive.com/lnk/1VY0rpwz

With 0.92 the gcode is ok. With 1.01 the gcode is wrong (even negative coordinates), with the new beta it fails with errors:
Traceback (most recent call last):
  File "laser.py", line 227, in <module>
    effect.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "laser.py", line 90, in effect
    curves = parse_string(etree.tostring(self.document))  # Parse an svg file into geometric curves
  File "C:\Program Files\Inkscape\share\inkscape\extensions\svg_to_gcode\svg_parser\_helper_functions.py", line 86, in parse_string
    return parse_root(root, canvas_height, transform, draw_hidden)
  File "C:\Program Files\Inkscape\share\inkscape\extensions\svg_to_gcode\svg_parser\_helper_functions.py", line 66, in parse_root
    curves.extend(parse_root(element, canvas_height, transform, False, visible))
  File "C:\Program Files\Inkscape\share\inkscape\extensions\svg_to_gcode\svg_parser\_helper_functions.py", line 62, in parse_root
    path = Path(element.attrib['d'], canvas_height, transform, transform)
KeyError: 'd'

uwefrank, 2021-02-12:
There was an error in the file. I have deleted it. Now it works with beta 2 but gcode is absolutly wrong. Negative coordinates!
Also as mentioned by ninef: teh laser on/off commands are missing.
The new link and some wrong gcode:
https://my.hidrive.com/lnk/UlYULs87

G1 F 3000 X 217.1 Y -630.6;
M3 S255;
G1 X 226.08740000000003 Y -627.8258000000001;
G1 X 234.85520000000002 Y -624.4384000000002;

nineff, 2021-02-12:
I made a small test file, and it shows strange paths on the top left (see attached screenshot).
as far as I can tell, these paths are not actually in the generated Gcode.

Feature suggestions:
*It would be nice to have the ability to see which G-code got generated from which path in the debug view (e.g. in the object properties description)
*the debug paths should have arrows indicating the move direction
*it would be great if the interface had the option to add/subtract "global" coordinate offsets. (e.g. my pen is about 40mm to the left of the machine center) 

[image]

PadLex, 2021-02-12:
@uwefrank I think the buggy behavior is caused by the transform attribute. I still need to add support for parsing transforms (a pretty important feature).

PadLex, 2021-02-12:
@nineff Great suggestions. I think I'll implement all three. Maybe it's time to create a proper feature tracker.

No idea what the strange paths are. Could you share the file?

nineff, 2021-02-13:
> 
> 
> @nineff Great suggestions. I think I'll implement all three. Maybe it's time to create a proper feature tracker.
> 
> No idea what the strange paths are. Could you share the file?

sure, here you go:
(i had to zip it because github does not allow svg files directly)
[Zeichnung.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/5975885/Zeichnung.zip)

PadLex, 2021-03-01:
I just published what might become the stable version if y'all don't find any bugs. (v2.0-beta.3 tag, still in the v2.0 release candidate).

- Added support for transformations. (Solves most bugs)
- Beautiful arrows
- Custom printer origin and bed size
- Translate and scale G-code
- A bunch of minor bug-fixes

Known issues:
- Elliptical arcs can act-out for certain edge cases 

Please try it out and let me know if you find any bugs or if you think it's ready to go stable. Thank you all for you're helping out!

PadLex, 2021-03-01:
@nineff Adding G-code to object descriptions is a bit of a mess to implement right now. I like the idea but it will have to come after the first stable release.

nineff, 2021-03-01:
> I just published what might become the stable version if y'all don't find any bugs. (v2.0-beta.3 tag, still in the v2.0 release candidate).
>
>     * Added support for transformations. (Solves most [...]

great, I will test it later today or tomorrow

nineff, 2021-03-01:
Update: It fails with the following error if i use a spiral:
```
C:\Program Files\Inkscape\share\inkscape\extensions\svg_to_gcode\compiler\_compiler.py:50: UserWarning: Compile with an empty body (no curves). Is this intentional?
  warnings.warn("Compile with an empty body (no curves). Is this intentional?")
```
Moreover, it then deletes the spiral from the file!
[spiral.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6060892/spiral.zip)

PadLex, 2021-03-01:
@nineff I tested it and the extension is actually working correctly in that case.  You accidentally placed the spiral in a debug layer so of course it was deleted and not drawn when you ran the extension.

nineff, 2021-03-01:
well thats explains it, but it should then be harder to accidentally do this ;)

nineff, 2021-03-01:
so now I did some more intensive testing:

* **BUG:** The resulting G-Code apparently only uses the feedrate for the travel speed and not the cutting speed. 
* **BUG:** I don't see any arrow-heads in the debug view, just red lines
* **Request:** please use the file-choosing dialog for the Output Directory/File and the custom header/footer code 
(see here https://wiki.inkscape.org/wiki/index.php?title=Extension_GUI_Reference#File_chooser)
* is there any reason why the Approximation Tolerance is entered as string and not as float?

nineff, 2021-03-17:
- **Request:** Color code different movement types in the debug view (e.g. cutting vs moving, potentially each pass a different color?)
- **Request:** Please let us define different commands for laser on/off again via the UI, not just the power level for one command.

j-lang, 2021-03-23:
> so now I did some more intensive testing:
> 
> * **BUG:** The resulting G-Code apparently only uses the feedrate for the travel speed and not the cutting speed.
>
Feedrate can be simlpy corrected by changing line 98/99 in _compiler.py of current release candidate to: 
           ```
 code = [self.interface.laser_off(), self.interface.set_movement_speed(self.movement_speed), self.interface.linear_move(start.x, start.y),self.interface.set_movement_speed(self.cutting_speed),
                    self.interface.set_laser_power(1)]
```

TehOcean, 2021-04-01:
Hi,

I'd just like to add, like @nineff I also use this extension for plotting so retaining the M03 and M05 commands would be very useful. 

@nineff: how are you modifying the g-code files to make them worth with your plotter?

PadLex, 2021-04-06:
Version 2.0 is out! Thank you all for your help finding bugs and features. Let me know if you encounter any bugs or have feature recommendations by creating a new issue. 

It's still missing the colored movement lines. If anyone wants to add them please do.

closed 2021-04-06 by PadLex