JTechPhotonics/J-Tech-Photonics-Laser-Tool #27: Output Jumbo sized

tychobarendse, 2021-02-05:
Hey,
I tested the plugin in Mac and checked the Gcode with a Gcode visualiser.
The paths generated are jumbo sized.
Am I not using the correct settings or is it a bug in the script?

PadLex, 2021-02-05:
That's odd. Try using the 2.0 beta version and let me know if it works better.

uwefrank, 2021-02-11:
I have the same issue. It only happens if I open an externel jpg, svg, dxf (that I tried). Same issue if I copy/paste the object If I draw something in Inkscape - no problem.

Neon22, 2021-02-12:
Looking through the code there is only one use of ```self.svg.unittouu```
Its possible that the doc units are being assumed and so the end result is not properly scaled.

uwefrank, 2021-02-12:
Attached a link to a file with a shape that does not work with the 1.01 Inksacpe J-Tech plugin.
https://my.hidrive.com/lnk/1VY0rpwz

The first lines of the 1.01 version: (wrong, even negative coordinates)
G1 F100.000000
G3 X259.2716 Y-611.0796 I-36.0412 J133.1766
G3 X294.5 Y-578. I-90.9258 J132.1304

The first lines of the 0.92 version (correct):
G1 F100.000000
G3 X112.1328 Y50.8541 I-9.5359 J35.2363
G3 X121.4536 Y59.6064 I-24.0574 J34.9595

PadLex, 2021-02-22:
@uwefrank Hey. I'm trying to verify if this bug was resolved in the 2.0 pre-release. Unfortunately the link broke. Could you share it again?

uwefrank, 2021-02-23:
Hi, I attach the file and the output gcode of 0.91 and 1.01.
[Jtech_test.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6027483/Jtech_test.zip)

PadLex, 2021-03-01:
Confirmed it's fixed in the latest version :)

closed 2021-03-01 by PadLex