JTechPhotonics/J-Tech-Photonics-Laser-Tool #30: Behaviour of path_depth

tommy3001, 2021-03-23:
Hallo. 
I tried the blending edge JTP-Laser-Tool_v2.0-beta3.zip today. 
I like the integration of SvgToGcode. 

But I am not lucky with the meaning of _path_depth_. It is not working in the for me expected way. In SvgToGcode, it is only added when  _passes_ is higher than 1.   It adds then after a complete svg graph cycle G1 Z-path_depth. In this way, it is not usable as a traveling height, nor as a correction when burning deeper in the material after a few cycles.  But that is an issue more related to SvgToGcode. I just added it here, to give the hint, that path_depth is relatively useless.

tommy3001, 2021-03-23:
Never mind. The post was mainly based on a misinterpretation of the code.

PadLex, 2021-04-06:
Should be fixed with the 2.0 release. Can you confirm?

closed 2021-04-07 by PadLex