JTechPhotonics/J-Tech-Photonics-Laser-Tool #33: Top-left origin flips image

bigtree908, 2021-04-06:
Using the top-left origin has the effect of flipping the image. I believe this is because the Y coordinates are still positive. If the origin is top left should that mean all the Y coordinates would then be negative? Either way when I run the script with the top left option then load the output into the sender the image is flip and the origin is still bottom left.

PadLex, 2021-04-07:
Technically it's behaving correctly, however, I just realized how confusing this is. When you use the top-left option, the origin is set to the top-left corner and the y-axis is flipped such that it increases as you go downwards. This may sound absurd, however it's actually really common, even SVG's work that way. Perhaps I could separate the option to invert the y-axis from the origin's position. What do you think?

bigtree908, 2021-04-07:
So I guess that means there is a difference between the origin and the 0,0 coordinate. Perhaps provide options for the position of both?

My machine homes to the top left. So the working space is the +X -Y region. But because every scrip I've experimented with outputs in the +X+Y region I have to manually move the head down and reset 0,0. I could invert the Y axis on my machine but then it wouldn't correspond with the preview of the sender.

PadLex, 2021-04-08:
I just separated the two options. Should be a lot more intuitive now. Let me know what you think of the changes :)

bigtree908, 2021-04-10:
I guess I'm confused what impact the 'origin' has on the output gcode. I ran a test with a simple diagonal line. Top-Left and Bottom-Left gave the same output. However, center had the expected result of shifting the 0,0 coordinate to the center,center of the design.

closed 2021-05-05 by PadLex