JTechPhotonics/J-Tech-Photonics-Laser-Tool #34: Desktop can't be selected as output folder

nineff, 2021-04-07:
Version 2.0 Release, Win 10 x64
Choosing the Desktop as output directory fails with the error message "Kein Object für Moniker" (No Object for Moniker)
This might as well be an Inkscape or GTK Bug, but I've only noticed it here.
Creating a new folder below the desktop ant choosing that works, though.

PadLex, 2021-04-07:
I think you're right, this does look like an Inscape bug.  However I can't reproduce it. Perhaps it has to do with your unique path structure?

nineff, 2021-04-09:
my path structure is very ordinary, in my case it is `C:\Users\Niko\Desktop`. But perhaps it only appears for localized Windows versions, since I'm using the German one?

PadLex, 2021-04-11:
Well that's odd. You might want to write a bug report on [Inkscape's official repo](https://gitlab.com/inkscape/inkscape/-/issues). Since it doesn't look like this extension caused the bug, I'll close this issue.

closed 2021-04-11 by PadLex