JTechPhotonics/J-Tech-Photonics-Laser-Tool #43: Compile with an empty body (no curves)

skyline2016, 2021-05-02:
I imported a hexagon shape to generate G code but the Gcode results is empty and returned "Compile with an empty body (no curves) Is it intensional?" I realized I cannot even use this tool to generate a simple rectangle. Is there anyway to fix it?

PadLex, 2021-05-05:
Should have been fixed with v2.3. Could you confirm?

skyline2016, 2021-05-05:
[image]

[image]
I freshly installed V2.3, but both the pattern and rectangle still give out the same result.

PadLex, 2021-05-05:
Did you remember to convert the shapes to paths? Select them all and go to Paths > Objects to Path

It's an important step because the plugin can only convert paths to gcode.

skyline2016, 2021-05-05:
Yes I know, I draw the shape with pen tool in illustrator and save it as SVG then import to inkscape. I have been image trace pictures in illustrator, expand them and save as SVG, then put in inkscape all successfully create Gcode. Just found out if I draw shape in inkscape directly the extension tool works fine. Would that be a problem from illustrator?

PadLex, 2021-05-05:
How odd. Could you attach one of the illustrator files that doesn't work? It would help me figure out what's going on.

skyline2016, 2021-05-05:
[pattern.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6427772/pattern.zip)
I included two svg in the zip file. Both are not working

PadLex, 2021-05-05:
Hmm. I can't reproduce the problem. Neither file contained any paths so I selected everything and converted every shape to a path. As you can see from the picture below, the gcode compiles correctly on my side.
[image]

Are you absolutely sure you selected everything before converting? If so, I have no idea what's going on. Try to convert all the shapes and then send me the files, there might be something wrong with Inkscape's "Object to Path" function.

skyline2016, 2021-05-05:
"Neither file contained any paths" is a big hint, may I ask what software you use to open SVG to check if there is path or not?
I am very sure the shape is identified as "path" in illustrator, since I use pen tool to draw it and it shows "path" when my cursor pointing it. I think it is a problem with inkscape that could not recognize it, when I import the svg into inkscape I select it to use "object to path" then the extension works well. It is weird that I don't need to use "object to path" when I import other svg into inkscape. Thanks for the help.
TLDR : apply "Object to path" in inkscape first no matter what.

closed 2021-05-05 by skyline2016

andraisr, 2023-10-03:
I am experiencing the same issue. I can create a drawing with the tools in inkscape. the gcode is created just fine. If I import a raster or svg file, I get this same issue. ("Compile with an empty body (no curves). Is this intentional?"). If the file is a raster i did Path > Trace Bitmap. Then verified image for additional lines with “Fill and Stroke”. Then I did "object to path, selected the image. then I went to the extension to create the gcode. Now if I draw an object with the tools within Inkscape everything works fine. Here is a silly additional question.... All this so I do not have to go to another program to process the the image and start engraving or cutting. This process creates a gcode file in the directory given. so now what, copy to a sd card and print from sd? Or does this extension add on have another interface to control the laser itself for printing?