JTechPhotonics/J-Tech-Photonics-Laser-Tool #46: Last cut always comes out wrong/problem with gcode gen

nmeurer, 2021-05-19:
Hi there,
I use the JTP tool for converting my 3d printer into a vinyl cutter.
However, I always encounter a flaw regarding the gcode generation.
The last cut/line/movement (which in this example is also the first one) is always off or not existent/gets ignored.
For this example, I choose to draw a simple star with Inkscape and plotted it's Gcode on my printer, which lead to the following result: [here are the pictures](https://imgur.com/a/Grz583I)

As you can clearly see, the last point of the star is clearly off.

Please don't get distracted by the red offset. Since the carriage which is holding the drag knife is a bit off from the center, I had to edit this in.

Best regards!

PadLex, 2021-05-19:
How odd. Could you attach the svg file and the gcode output?

nmeurer, 2021-05-20:
Sure, but github doesn't allow uploading .svg and .gcode files.
Please download the corresponding files and remove the .txt ending.

Cheers
[EnderTemplate.svg.txt](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6516597/EnderTemplate.svg.txt)
[output.gcode.txt](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6516598/output.gcode.txt)

PadLex, 2021-05-21:
Love your workaround. Why is GitHub even policing file extensions though?

Anyway, I plugged your gcode into a simulator and the star looks symmetrical. However, it also raised an error because the gcode told it to move (G1 Z3) before specifying a movement speed (F1000). I'm not sure how that happened. Did you add the G1 Z3 lines manually?
[image]

I've attached a corrected output that doesn't raise any errors in the simulator. Does it still create asymmetries when cut? Regardless, I have no clue what's going on... Could it be your machine? I just can't imagine how a machine can consistently cut everything but the last two lines correctly. Yet the gcode looks fine.

[clean_output.gcode.txt](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6521860/clean_output.gcode.txt)

nmeurer, 2021-05-22:
I replaced the tool on / tool off command by G1 Z0 and G1 Z3 to lower my drag knife into the material that I want to cut, since there is no way to make it work with the traditional laser commands. Although the JTP laser tool seems to focus on lasers, it of course is quiete popular for plotters and vinyl plotters as well. 
I'm going to try out your fixed gcode and report back what happened.

Cheers

PadLex, 2021-05-22:
BTW, check out this [issue](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/issues/47). With the latest version of the extension you can directly specify the laser on/off command.

nmeurer, 2021-06-01:
Hi there,
sorry for my late reply but I didn't have time to test the gcode until today.
Unfortunately the problem still exists.

Cheers

PadLex, 2021-06-09:
How bizarre. Have you tested it with a different gcode generator?

closed 2022-02-19 by PadLex