JTechPhotonics/J-Tech-Photonics-Laser-Tool #47: Can you add an check box to remove/add the S0 from laser off cmd?

murilobaliego, 2021-05-20:
I'm trying to use your tool to work not with laser but as a drawing using the cmds M3 S90 to down the pen and M5 to lift the pen. However the S0 that is added hard coded on laser off cmd I need to handle it either by changing the code .py or by find/replace the M5 S0 with M5 only. Would be nice to have in your interface an option to add or not this S0 on laser off command.

PadLex, 2021-05-21:
You bring up a good point. I updated the extension so now It'll just use whatever command you give it for powering the tool on and off. Ie, you can just specify M3 S90 without pointless complications. Let me know if it solves your issue.

murilobaliego, 2021-05-24:
Thanks for the updated code Alex, I have tested it successfully during this past weekend, here is [one video](https://www.youtube.com/watch?v=4-3r65-LMkQ) that I made with your extension, it worked perfectly. May I suggest to you to rename your extension as it is not only for Laser Tool, it is also for any kind of CNC drawing. ;)
Great job!

PadLex, 2021-05-25:
Wow, that picture of Ronaldo came out really well! I'm glad the extension is being used for more then just laser engraving :)

closed 2021-05-25 by PadLex