JTechPhotonics/J-Tech-Photonics-Laser-Tool #49: Cannot engrave (just cutting) with grbl and inkscape

dekiakbar, 2021-06-09:
I have build A laser CNC running arduino with :
GRBL 1.1h
Inkscape v1.0.2
J-Tech-Photonics-Laser-Tool v2.4

BTW I user archlinux (I think this is not OS issue)
everithyng is good but how to do engrave using this plugin ? I have read the doc here but still cannot engrave https://jtechphotonics.com/?page_id=2012

PadLex, 2021-06-09:
Did something go wrong or is process unclear? I've been putting off working on the docs, I think they're a little confusing.

dekiakbar, 2021-06-19:
yeah the docs id little confusing, I think it must be updated, when I engrave a (i.e star image), only the outline of the star image is engraved. the plugin generated gcode without filling the area of star image. can you reproduce it on your own machine ?

PadLex, 2021-06-19:
Yep. This extension specifically converts svg paths to gcode for engraving and cutting. In other words, it draws lines. To fill in an area I recommend you look for png/jpeg to gcode engraving projects.

dekiakbar, 2021-06-19:
aah okay thanks @PadLex , I will close this issue 👍🏻

closed 2021-06-19 by dekiakbar