JTechPhotonics/J-Tech-Photonics-Laser-Tool #50: Offset in y axis

agryson, 2021-06-19:
Using Inkscape 1.1 I get a warning while running about transforms (maybe related) and upon generation notice that the gcode is offset downwards by about a centimetre.

The points in this screen shot should be at each corner of the canvas for example.

[image]

agryson, 2021-06-19:
The offset is precisely 11.25mm. I can work around this by manually remembering to offset prior to gcoding, but naturally a proper fix (or info on what I may be doing wrong?) would be helpful.

PadLex, 2021-06-19:
Welcome!

It looks like you're using an outdated version of the extension. Try installing the latest 2.x version ;)

agryson, 2021-06-19:
🤦 Thank you, you're right, I originally downloaded from https://jtechphotonics.com/?page_id=1980 so was on V1.1_beta.
Jus ttested and it works _perfectly_.
Apologies and thanks.

closed 2021-06-19 by agryson