JTechPhotonics/J-Tech-Photonics-Laser-Tool #52: Compile with an empty body

zorgoban, 2021-07-22:
Hi!

Inkscape 1.1 (c68e22c387, 2021-05-23)
2.4 stable release
This very simple svg File. Github doesn't accept SVG so i just renamed it to XLS.
[drawing.xls](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6864317/drawing.xls)

I get the following error message:

C:\Users\remo\AppData\Roaming\inkscape\extensions\laser\svg_to_gcode\compiler\_compiler.py:63: UserWarning: Compile with an empty body (no curves). Is this intentional?
  warnings.warn("Compile with an empty body (no curves). Is this intentional?")

Many Greetings!
Remo

zorgoban, 2021-07-22:
Solved it with converting to path. I don't have to understand why this is necessary, right?

closed 2021-07-22 by zorgoban

andraisr, 2023-10-02:
How exactly did you get this error resolved? I am attempting to load the laser tool extension nad seeing the same error.

zorgoban, 2023-10-03:
Hi! This was 2 years ago. As I wrote: Convert to path. That is somewhere in Inkscape. Didn't use Inkscape nor this tool for the past two years.