JTechPhotonics/J-Tech-Photonics-Laser-Tool #53: top-left origin always bottom-left in gcode

sninaber, 2021-08-09:
i like to use top-left as machine origin. mainly because inkscape also uses this as coordinate system and i can adjust objects accordingly with ctrl-shift-x to my measurements. also, this way the grids of inkscape en my gcode sender (bcnc) line up.

i installed this extension yesterday but couldn't change the origin by means of the option provided in the inx-file, the gcode output always originated bottom-left.

have made 2 changes which seem to correct this for me, included in attachment. please review.
[laser.txt](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/files/6956579/laser.txt)

odaki, 2021-08-15:
Are you using the latest code from master?
With my little patch, left-top is working as intended in my environment.

https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/commit/1d17942550b36929ca55a1fa60680f077fd016bc

sninaber, 2021-08-15:
hi odaki, thanks for your reply!
yes, i cloned from master, but will try again next week (away from home now). will let you know if problem persists.

steven.

1d17942<https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/commit/1d17942550b36929ca55a1fa60680f077fd016bc>