JTechPhotonics/J-Tech-Photonics-Laser-Tool #55: Curves are being generated with a huge number of steps on GCode

rafzan, 2021-08-17:
Debug Image:
[image]

Here a piece of the GCode:

`G1 X42.968332 Y36.488927;
G1 X43.132117 Y36.550031;
G1 X43.298404 Y36.557583;
G1 X43.464269 Y36.518020;
G1 X43.626788 Y36.437775;
G1 X43.783037 Y36.323284;
G1 X43.930094 Y36.180981;`

As you can see, this is a sequence of lines instead of a proper curve. 
I'm trying to draw a simple standard spiral from illustrator toolbox.

PadLex, 2021-08-18:
That's actually how the extension has always worked. At the end of the day CNC machines only move in straight lines, even when you tell them to perform an arc they're just performing a thousand little straight line movements.

In the latest version you can actually see the lines represented as arrows. If you don't want to see them you can disable debug arrows. If you're looking for a smoother curve, you can decrease the approximation tolerance. Conversely, by increasing the tolerance you'll get fewer lines and a smaller gcode file.

rafzan, 2021-08-18:
Thanks @PadLex, sorry for my lack of attention.

The problem I was having is related to the **dwell time before moving**. I need to set a **power-on delay**, like in the old extension. This dwell configuration is making my plotter stop for every line change on a chain.

rafzan, 2021-08-18:
Thanks @PadLex, sorry for my lack of attention.

Actually, decreasing the approximation tolerance yielded better results. My plotter was chopping too much with the default 0.01 setting, but no matter the setting I use, it's never as smooth as arc functions.

closed 2021-08-18 by rafzan