JTechPhotonics/J-Tech-Photonics-Laser-Tool #56: ModuleNotFoundError: No module named 'svg_to_gcode'

dfgianni, 2021-08-18:
hi, i have this error with the J-Tech-Photonics-Laser-Tool. it is installed on a 1.0.2-2 inkskape version. how i can resolve this problem?

dfgianni, 2021-08-18:
the J-Tech-Photonics-Laser-Tool version is 2.4. many thanks

PadLex, 2021-08-18:
You could downgrade to a previous version of the extension or install the latest version of Inkscape. If neither works make sure to post the error so we can help you better.

dfgianni, 2021-08-18:
upgradeded inkscape to the last version 1.1 but i have always the same problem in attached:

Traceback (most recent call last):
  File "laser.py", line 7, in <module>
    from svg_to_gcode.svg_parser import parse_root, Transformation, debug_methods
ModuleNotFoundError: No module named 'svg_to_gcode'

PadLex, 2021-08-18:
The error is saying that the svg_to_gcode folder isn't in the same directory as the laser.py file. When you install the extension you need to make sure you copy over both the svg_to_gcode folder and the laser.py/laser.inx files.

dfgianni, 2021-08-18:
oh, i'm sorry. wich is the download file of the svg_to_gcode? can you help to to install it correct? it seems that i need to install it with pip or i can just copy files?

dfgianni, 2021-08-18:
[image]

superfluent, 2021-12-26:
For n00bs like me:
- Make sure you got Python installed and added tp PATH
- start cmd
- cd to location of setup.py
- execute "setup.py install"
(Installs successfully)

Unluciky I remain with the following errors: 
[image]

PadLex, 2022-02-19:
How odd, if it isn't too late, could you post a picture of your directory structure after installing the library? It's very important that the laser.py and laser.inx files be in the same location as the svg_to_gcode folder.

closed 2023-08-18 by PadLex

andraisr, 2023-10-02:
Thanks for this info. I surmise that the contents of the svgtogcode zip file be saved into the laser folder.