JTechPhotonics/J-Tech-Photonics-Laser-Tool #57: No more "power-on delay" setting?

rafzan, 2021-08-18:
Hello again!

I use this extension with my [4xiDraw plotter](https://www.instructables.com/4xiDraw/). In my opinion it's the best software option by far.

The problem is that the pen-up/down mechanism is controlled by a micro-servo which is orders of magnitude slower than any laser device. The 0.92 extension had a nice power-on/off delay setting which stopped the carriage, waited for the pen to go up/down and then resumed the walking. Without this setting, this happens:

[image]

The **dwell time before moving** does not fill this condition. It adds the G4 commands in bad places for my situation.
A solution was adding `G4 P0.5` after m3 and m5 commands directly to the code. Gcode doesn't seem to recognize them when I add directly to the **Tool Power/Off Command** inputs, as they have to go inline in the code.

I don't know if this extension is intended for plotter usage, but it fits my needs really well apart for this. Should I suggest a solution and make a pull request?

superfluent, 2021-12-26:
Same problem for me (as probably any other drawbot user) so +1

PadLex, 2022-02-19:
I think someone implemented that feature in the main branch but I haven't tested it enough to create a new release. If anyone is curious you could try cloning the repo and see if it works for you. Otherwise the older releases are still there.

Corgano, 2022-09-26:
Crap I ran into this same issue, was hoping someone had a solution :/

Jodaille, 2023-02-06:
Hello,
it seems ok on main branch.

I have just cloned the repository, and put laser.inx  laser.py files
in /usr/share/inkscape/extensions/laser directory

with latest version of svg_to_gcode:
https://github.com/PadLex/SvgToGcode/tree/master/svg_to_gcode

[image]

Gcode output sample:
`G90;
M5;
M5;
G21;
G4 P0.5
M5;
G1 F3000.0 X42.788652 Y123.524713;
M3 S3000;
G1 F750.0 X34.239848 Y100.590852;
G1 X51.368658 Y100.590852;
G1 X42.788652 Y123.524713;
G4 P0.5
M5;`