JTechPhotonics/J-Tech-Photonics-Laser-Tool #60: Error in Output Directory! message

BCChad, 2021-11-09:
On setting up the laser job and pressing "Apply" I get the simple error message "Error in Output Directory!"
I have tried several variations of using your button to select the directory, or  typing it in directly.  Is there a limit to how deeply nested the subdirectory is? (I don't think so as I tested various alternatives to work around this.  NO JOY!).
Please advise.  It is your most recent version, from 9/3/21, downloaded today 11/8/2021

PadLex, 2021-11-10:
Thank you for pointing that out. I'm still not sure why it crashed but I'll look into it.

BCChad, 2021-11-10:
Hey Alex, sorry - it was not a crash, just an undesirable outcome. But my "Aha" explains that I figured out that it was my fault by not supplying the filename extension.  My suggestion - if you could provide a prompt to clarify this for simple-minded users that would be great; or possibly supply a default ".txt" extension if the user fails to provide one.

PadLex, 2022-02-19:
Good idea, I'll tag this issues in the meantime.

closed 2022-02-19 by PadLex