JTechPhotonics/J-Tech-Photonics-Laser-Tool #63: Exception when start

sstefanov, 2021-11-17:
When started I receive following:

Traceback (most recent call last):
  File "/mnt/sda2/sstefanov/.config/inkscape/extensions/laser.py", line 308, in <module>
    effect = GcodeExtension()
  File "/mnt/sda2/sstefanov/.config/inkscape/extensions/laser.py", line 44, in __init__
    EffectExtension.__init__(self)
  File "/usr/share/inkscape/extensions/inkex/base.py", line 304, in __init__
    super().__init__()
  File "/usr/share/inkscape/extensions/inkex/base.py", line 64, in __init__
    self.add_arguments(self.arg_parser)
  File "/mnt/sda2/sstefanov/.config/inkscape/extensions/laser.py", line 271, in add_arguments
    arguments = self.read_arguments()
  File "/mnt/sda2/sstefanov/.config/inkscape/extensions/laser.py", line 296, in read_arguments
    arguments.append({"name": name, "type": types[arg_type]})
KeyError: 'boolean'

I think line 19 in laser.inx must be like this:
<param name="filename_dynamic" type="bool" gui-text="Reuse SVG filename for output">false</param>
instead of:
<param name="filename_dynamic" type="boolean" gui-text="Reuse SVG filename for output">false</param>

PadLex, 2021-11-17:
According to [the Wiki] (https://wiki.inkscape.org/wiki/Extensions:_INX_widgets_and_parameters#:~:text=Result-,boolsince%C2%A0,-1.0) it's been changed to bool since v1.0.

Are you sure you are using the most up-to-date version of the extension?

sstefanov, 2021-11-17:
Yes, check line 19 here:
https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/blob/master/laser/laser.inx

PadLex, 2021-11-17:
My bad, I meant to ask if you're using the latest version of Inkscape. Versions below 1.0 require the use of `boolean` insted of `bool`

sstefanov, 2021-11-18:
My Inkscape version is 1.1.
In all cases I think there must be the same types in the file.
When I changed "boolean" to "bool" in .inx file this passes fine.