JTechPhotonics/J-Tech-Photonics-Laser-Tool #65: Enhancement Request - Specify start point

joeynuggetz, 2022-01-25:
Hi, first of all, I really love your extension... That's what she said.  Anyway, would be nice if you could add a few more options like being able to specify where the beginning of the cut will start. I'm using this as a plotter but imagine would have a similar need if I was laser engraving something.  Lets face it, drawing on paper is sort of useless. The real value to me is being able to plot onto something other than paper which requires that I set the laser on and off to positions higher up in the Z axis so I can plot onto physical objects. 

Is there a way to know where the drawing/cutting or whatever will begin? Without having to figure out down to the mm where to place my object, I set the gcode to pause after reaching the first line of gcode. I then position the right side of my work just under the pen and after the countdown it continues but the drawing seemed to move off my object which tells me it doesn't start at the point closest to the home point all the time.

PadLex, 2022-02-19:
Sorry for the late reply. I like your suggestion but I don't have the time to implement it my self (I'm struggling to reply to issues as is :/). Perhaps you could give it a try and create a pull request. Otherwise I'd just edit the GCODE by hand, it's a lame solution but if you only want to edit the start and end it shouldn't be too hard.

itspacchu, 2022-04-20:
Maybe not a starting point , but Something like offset would be cool for those using their 3D printers with a mounted pen,
I might try working on the offset system.