JTechPhotonics/J-Tech-Photonics-Laser-Tool #67: Lot of useless movements

Carlsans, 2022-02-15:
I'm using this extension for generating gcode for a plotter that I've designed. The code generation is much faster with the last version but when printing there is a lot of useless no print movement. Because of that, I will use older versions.

PadLex, 2022-02-19:
If you have the time, could you attach a picture of the GCODE traces for the old and new versions? It would help me understand the difference.

Carlsans, 2022-02-20:
communauty.gcode
<https://drive.google.com/file/d/1ZBbAWcw6KZyDO2gDLBiqre2WSaRB28dG/view?usp=drive_web>
 old.gcode
<https://drive.google.com/file/d/1wxcBeTerv6XZNgWNvQf1oxSTgYPAC6oW/view?usp=drive_web>
I've sent a file (not the same image..) from the old version and one from
the communauty version. You can use Gcode viewer to visualize :
http://jherrm.com/gcode-viewer/
Thanks a lot for your work !
Carl

PadLex, 2022-02-22:
I'm afraid Drive is giving me an access denied error