JTechPhotonics/J-Tech-Photonics-Laser-Tool #68: Gcode issues

BallieK69, 2022-02-19:
Hi
When trying to make a simple design with an oval with text inside then convert to gcode the result on GRBL screen shows the text out of place. below is the message after converting. What is it that is going wrong.

C:\Program Files\Inkscape\share\inkscape\extensions\laser.py:928: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  t = simpletransform.parseTransform(t)
C:\Program Files\Inkscape\share\inkscape\extensions\laser.py:929: DeprecationWarning: simpletransform.composeTransform -> Transform(M1) * Transform(M2)
  trans = simpletransform.composeTransform(t, trans) if trans != [] else t
C:\Program Files\Inkscape\share\inkscape\extensions\laser.py:938: DeprecationWarning: simpletransform.applyTransformToPath -> Path(path).transform(mat)
  simpletransform.applyTransformToPath(trans, csp)

PadLex, 2022-02-19:
The deprecation warnings shouldn't matter too much. Could you attach a picture of the output's traces?

BallieK69, 2022-02-19:
This is how it starts out and ends up at GRBL side. Using latest Inkscape and windows 11.

[image]
[image]
[image]

BallieK69, 2022-02-20:
Is there anything else needed?

PadLex, 2022-02-22:
I'm not able to reproduce the bug with Inkscape 1.1.1 and the latest release of this extension [JTP-Laser-Tool_v2.4.zip](https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/releases/download/v2.4/JTP-Laser-Tool_v2.4.zip)76.2

Could you attach the buggy svg file? Also, judging from the size of the arrow traces, is there any chance you're using a release older than 2.0?

BallieK69, 2022-02-22:
Yip. Latest Inkscape 1.1, J Tech 2.4 and GRBL 4.7.1
Looks like if the job is saved in "Optimized svg" then ok...