JTechPhotonics/J-Tech-Photonics-Laser-Tool #69: KeyError: 'boolean'

Paulink-sys, 2022-03-12:
Hi, I am trying (Win10) to use Inkscape 1.1.2 (0a00cf5339, 2022-02-04, custom) with the latest version of https://github.com/PadLex/SvgToGcode. When doing so, I get the following error message:

What can I try?

```
Traceback (most recent call last):
  File "C:\Users\Paul Müller\AppData\Roaming\inkscape\extensions\laser\laser.py", line 302, in <module>
    effect = GcodeExtension()
  File "C:\Users\Paul Müller\AppData\Roaming\inkscape\extensions\laser\laser.py", line 41, in __init__
    EffectExtension.__init__(self)
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 304, in __init__
    super().__init__()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 64, in __init__
    self.add_arguments(self.arg_parser)
  File "C:\Users\Paul Müller\AppData\Roaming\inkscape\extensions\laser\laser.py", line 268, in add_arguments
    arguments = self.read_arguments()
  File "C:\Users\Paul Müller\AppData\Roaming\inkscape\extensions\laser\laser.py", line 293, in read_arguments
    arguments.append({"name": name, "type": types[arg_type]})
KeyError: 'boolean'

```

samuelverburg, 2022-03-23:
Got the exact same one before! Installed the 1.1 version of the J- Tech tool, which worked, but misses some nice features...
I installed the lastest J-Tech tool (2.4) again (but this time through the releases zip) and somehow I am now getting this error. It looks similar so I figured I'd share here. Perhaps they stem from the same underlying bug that needs to be patched.
`Traceback (most recent call last):
  File "C:\Program Files\Inkscape\share\inkscape\extensions\laser.py", line 290, in <module>
    effect.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 131, in run
    self.save_raw(self.effect())
  File "C:\Program Files\Inkscape\share\inkscape\extensions\laser.py", line 63, in effect
    if os.path.isfile(self.options.header_path):
  File "C:\Program Files\Inkscape\lib\python3.9\genericpath.py", line 30, in isfile
    st = os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
`

PadLex, 2022-03-24:
@Paulink-sys  The KeyError: 'boolean' bug should only occur if you use the main branch. I suggest using the 2.4 stable release instead.

PadLex, 2022-03-24:
@samuelverburg I'm unable to reproduce your error using Inkscape 1.1.1 and this extension's 2.4 stable release. I'm using my downloads as the output directory and the filename is called output.gcode. Judging from your stack-trace you are probably inadvertently using an older version of the extension because the line numbers don't correspond to the latest code. I'd recommend checking both your Inkscape extension folder and user extension folder to look for outdated versions of this extension.

odaki, 2022-04-17:
Hello PadLex.

#62 does not mesh with what I fixed in #54.
#62 should also use the type "bool", not "boolean".
Please correct.

from:
https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/blob/b7f55021e1fb2695f99d462f188507138f934ae4/laser/laser.inx#L19

to:
```xml
            <param name="filename_dynamic" type="bool" gui-text="Reuse SVG filename for output">false</param>
```

Thank you.

closed 2022-04-20 by PadLex

odaki, 2022-04-20:
@PadLex  Thanks for the correction.
If the policy is to unify to boolean instead of bool, please modify laser.py.

As it is now, it causes an error.
[image]

Please change laser.py
from:
https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/blob/b7f55021e1fb2695f99d462f188507138f934ae4/laser/laser.py#L291
to:
```python
            types = {"int": int, "float": float, "boolean": Boolean, "string": str, "optiongroup": str, "path": str}
```

Thank you.

PadLex, 2022-04-20:
@odaki thank you for pointing that out. I got confused and flipped them. Just pushed the fix.

commit 1a2f40ce (2022-04-20): Fix #69