JTechPhotonics/J-Tech-Photonics-Laser-Tool #70: Error creating gcode

dgg006, 2022-03-31:
I have Windows 10 x64.
Using the latest version of Inkscape and the latest stable version of the plugin.
```
Traceback (most recent call last):
  File "C:\Users\ServiCell\AppData\Roaming\inkscape\extensions\community.jtechphotonics.com\laser\laser.py", line 290, in <module>
    effect.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 131, in run
    self.save_raw(self.effect())
  File "C:\Users\ServiCell\AppData\Roaming\inkscape\extensions\community.jtechphotonics.com\laser\laser.py", line 63, in effect
    if os.path.isfile(self.options.header_path):
  File "C:\Program Files\Inkscape\lib\python3.9\genericpath.py", line 30, in isfile
    st = os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
```

CJRN-cst, 2022-04-01:
Windows 10 x64.
Beta version don't work as well.
Gcode way out of whack. using Laser-accuracy-test-file.svg file
C:\Users\compu\AppData\Roaming\inkscape\extensions\laser.py:928: DeprecationWarning: simpletransform.parseTransform -> Transform(str).matrix
  t = simpletransform.parseTransform(t)
C:\Users\compu\AppData\Roaming\inkscape\extensions\laser.py:929: DeprecationWarning: simpletransform.composeTransform -> Transform(M1) * Transform(M2)
  trans = simpletransform.composeTransform(t, trans) if trans != [] else t
C:\Users\compu\AppData\Roaming\inkscape\extensions\laser.py:938: DeprecationWarning: simpletransform.applyTransformToPath -> Path(path).transform(mat)
  simpletransform.applyTransformToPath(trans, csp)

CJRN-cst, 2022-04-02:
update. did a clean install on Inkscape 0.92 and associated plugin V2_2 for Inkscape 0.92seems to have fixed issues.

tfindlay-au, 2022-04-07:
Windows 10 x64 build 19044
Python 3.9.2 installed

Still an issue with current stable release of Inkscape 1.1
JTP Laser Tool version 2.4

```
Traceback (most recent call last):
  File "C:\Users\Timothy\AppData\Roaming\inkscape\extensions\laser\laser.py", line 290, in <module>
    effect.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 131, in run
    self.save_raw(self.effect())
  File "C:\Users\Timothy\AppData\Roaming\inkscape\extensions\laser\laser.py", line 63, in effect
    if os.path.isfile(self.options.header_path):
  File "C:\Program Files\Inkscape\lib\python3.9\genericpath.py", line 30, in isfile
    st = os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
```

A simple check before the call may help.

itspacchu, 2022-04-19:
@dgg006 @tfindlay-au 
For a simple hacky fix you can create an empty file and just set custom gcode paths to empty file and it seemed to work
somehow its not checking for empty path.. I'll probably try to fix it.
[image]

closed 2022-04-20 by PadLex