JTechPhotonics/J-Tech-Photonics-Laser-Tool #72: Definition of origin is bugged?

AlainFarigoule, 2022-05-08:
When you have a machine (as an example Creality Endre3Pro) with its coordinates originating at bottom-left corner, you must use the same geometrical reference (if you want to remain sane, that is).
The latest version of Inkscape (V 1.1.1 (3bf5ae0d25), 2021-09-20) has changed its coordinate system as compared to previous versions 0.92.
However, although the default is top-left origin, you still can specify a bottom-left one by going to Edit\Preferences\Interface\untick the box "Origin at upper left with y-axis pointing down"
The corresponding version of JTech Photonic Laser Tool is v2.4, which accordingly has the option to choose top-left or bottom-left corner. This is specified in Extensions\..\JTech Community Laser Tool\Cordinate System & Transformations choosing list
This, I believe, has a bug.
Whatever you specify, the JTech tool will install a top-left origin. This you can see after the routine has run, since the origin is marked on the "debug reference points" layer just created. If you simulate your g-code you will get a confirmation of that.
This results in a g-code that cannot be executed by your machine because it has negative Y values, which are refused by the machine as they are outside its bed!
This is apparently not new, if I refer to the GitHub https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/releases with post #33). A reversal of the Y-axis appears to have been introduced in an attempt to solve the problem. This creates a symetry with respect to the x-axis, which is very different from the required displacement along Y-axis. 

Until a revised version is available, I apply the following geometric "patch".
Once my design is complete in Inkscape, I read the Page Height from File\Document Properties\Custom Size\Height.
Then I move the design upwards: Select the whole design and replace its Y value by Y + PageHeight. The design moves up, out of the designated Page, but is correctly located with respect to the (future JTech) origin.
Then, as before Select the design, Path\Object to path and Extension\..\Jtect Community Laser tool\provide parameters + Apply.
The digital version would be to change all Y values of the object file by Y + PageHeight. 
My competency in python is not adequate to do that; can someone with correct competency check this and apply correction, please

Knighthawk1970, 2022-06-10:
I am also having this issue. No matter what I choose, origin is always top-left. Anyone working on this?

wtarreau, 2022-09-04:
I think I'm facing the same issue, the plugin is non-functional anymore. The output starts with a large Y offset (or sometimes negative values of X, for an unknown reason) and it seems to me it's reversed no matter whether I click the "invert Y axis" or not. Also I don't understand what the "bed X and Y" values are for, maybe a security for some devices ? It seems to participate to calculating the wrong values for me and they appeared to be "less wrong" (i.e X no more negative) when I set 0 to both, but that looks really odd. I feel like I'll have to reinstall my PC with the previous OS and packages if I still want to use my engraver as I have no idea what's wrong there and can't read (let alone write) python :-(  I'm tempted to try to revert to plugin version 1.1 which didn't rely on this new lib, but I don't know if it will be compatible with inkscape 1.2.

sergeolkhovik, 2023-05-06:
After updating to Ubuntu 23.04 (and Inkscape 1.2) I've got similar problems with the master code. X is OK, but every time Y shifted (and bed size set affects this as well).
After several tries I discovered svg2gcode (https://github.com/sameer/svg2gcode) the works perfectly.