JTechPhotonics/J-Tech-Photonics-Laser-Tool #80: Inkscape 1.2.2 Not running Jtech Photonics or Endurance Extensions, even though they show in the list

JayC2469, 2023-01-31:
Hi All
I am having issues getting Anything to happen when selecting either the Jtech Photonics and/or the Endurance Lasers extensions to generate Gcode for Laser Etching or cutting. The extensions all appear in the list but the current latest version of Inkscape I am using (1.2.2), is not playing ball, with regards to the Gcode Generating Extensions mentioned.
I have used Inkscape previously, in the past on version 0.98 using Jtech Photonics Gcode plugin with varying degrees of success eventually. 
This is a clean install of the latest version I am using, on a relatively clean installation of Windows 10 Pro, Build No. 19044.2486
Any help would be  Greatly appreciated!!

JGLSoft, 2023-02-03:
Traceback (most recent call last):
  File "%UserProfile%\AppData\Roaming\inkscape\extensions\laser.py", line 312, in <module>
    effect.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
    self.save_raw(self.effect())
  File "%UserProfile%\AppData\Roaming\inkscape\extensions\laser.py", line 71, in effect
    filename, extension = output_path.split('.')
ValueError: too many values to unpack (expected 2)

Jodaille, 2023-03-30:
Hello,
which version of python do you use ?
have you tried to install plugin using **main** branch?

cf: https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/issues/57#issuecomment-1419882285

sergeolkhovik, 2023-05-06:
After updating to Ubuntu 23.04 (and Inkscape 1.2) I've got similar problems but switching to master branch resolved this problem.

Doubletop12, 2023-08-15:
@Jodaille @sergeolkhovik Does this relate to my issue #83? What do you mean by "main branch" or "master branch"?

sergeolkhovik, 2023-08-16:
@Doubletop12 well, in my case it generated an error when running, if I remember correctly. I didn't have any issue with greyed menu item. Yes, I downloaded files directly from main/master branch (previously that was some release, but I don't remember which one). Anyway that didn't help me I switched to svg2gcode (see: https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/issues/72) and all is fine now. That tool runs much faster and doesn't have issues with some fonts.

Doubletop12, 2023-08-17:
@sergeolkhovik Thank you I do think I have a diffrent problem. Maybe compatibility with V1.3 of Inkscape. 

I now understand what you mean by main branch or master branch, you are refering to the version of the laser plugin not Inkscape.

Pete

sergeolkhovik, 2023-08-17:
@Doubletop12 yes, exactly, the main/master branch of the plugin code.

ILYB, 2024-02-21:
I'm not able to generate g-code using the J-Tech Community Laser Tool Extension either. The below error comes up when I hit apply after selecting Object to Path and then opening the extension. I'm running latest version of Inkscape 1.3.2 and down loaded the latest J-Tech-Photonics-Laser-Tool-master extracted and copied the Laser subfolder and then pasted it in the extension folder as shown below. New to Inkscape and use of this extension so any suggestions would be appreciated.

Line 7 in laser.py reads from svg_to_gcode.svg_parser import parse_root, Transformation, debug_methods

laser.inx has the below script included (I do know if this is pertinent but appears to contain similar reference).
<script>
        <command location="inx" interpreter="python">laser.py</command>

C:\Program Files\Inkscape\share\inkscape\extensions\laser
contains laser.inx and laser.py

Error message

Traceback (most recent call last):
  File "C:\Program Files\Inkscape\share\inkscape\extensions\laser\laser.py", line 7, in <module>
    from svg_to_gcode.svg_parser import parse_root, Transformation, debug_methods
ModuleNotFoundError: No module named 'svg_to_gcode'

Jodaille, 2024-02-21:
Hello,
the error No module named 'svg_to_gcode' means that svg-to-gcode module could not be found (could be issue with python path or not installed) https://pypi.org/project/svg-to-gcode/

I don't know about windows, but it should be installed with **pip**

`pip install svg-to-gcode`