JTechPhotonics/J-Tech-Photonics-Laser-Tool #83: Extension greyed out

Doubletop12, 2023-08-15:
Inkscape 1.3 and Laser Tool 1.0 and Win 10

The extension "Generate Lase Gcode" appears in the list but the "J-Tech-Photonics-Laser-Tool " is greyed out and nothing happens when it is clicked.

Extension is located in "C:\Users\Pete\AppData\Roaming\inkscape\extensions\J-Tech-Photonics-Laser-Tool-1.0-beta_ink0.9"

Pete

Doubletop12, 2023-08-17:
I have now removed Inkscape 1.3 and replaced it with 1.2 the greyed out laser tool menu item is now not greyed out. However the laser tool dialogue box still doesn't appear.

PadLex, 2023-08-17:
Hi, are you sure you're using the extension's latest release? I think file path should look more like `C:\Users\Pete\AppData\Roaming\inkscape\extensions\laser`.

Doubletop12, 2023-08-18:
@PadLex that is what the instructions say but when you unzip the files to the location you get
"C:\Users\Pete\AppData\Roaming\inkscape\extensions\J-Tech-Photonics-Laser-Tool-1.0-beta_ink0.9" the menu item is "J-Tech-Photonics-Laser-Tool-1.0-beta_ink0.9"
if you rename the folder to

"C:\Users\Pete\AppData\Roaming\inkscape\extensions\laser" you still get "J-Tech-Photonics-Laser-Tool-1.0-beta_ink0.9" as the menu item

I have now tried it with 1.02 and the menu item is greyed out so I have gone back to 1.2 where the menu item is not greyed out and with the folder name chnged to laser. Still the extension doesn't open

Doubletop12, 2023-08-18:
I have now found the debug bat file and get this at startup. However Inkscape starts and appear to work but the menu laser menu item does not work but no further debug messages
```
Launching Inkscape, please wait...
warning: onecore\base\appmodel\runtime\winrt\lib\windows.applicationmodel.package.cpp(1008)\Windows.ApplicationModel.dll!00007FF9E508F63F: (caller: 00007FF9E507C5BC) ReturnHr(1) tid(35d8) 80070002 The system cannot find the file specified.
warning: onecore\base\appmodel\runtime\winrt\lib\windows.applicationmodel.package.cpp(512)\Windows.ApplicationModel.dll!00007FF9E508F4B1: (caller: 00007FF9ABC1BDE3) ReturnHr(2) tid(35d8) 80070002 The system cannot find the file specified.
```

PadLex, 2023-08-18:
Could the extension be nested in another folder? What's the absolute path to `laser.py` and the `svg_to_gcode` folder?

Doubletop12, 2023-08-26:
@PadLex  Thanks that got me looking and I don't have an SVG-to-Gcode. appart from a casual mention in a requirements.txt file their is no reference to needing it in the README.mdso I hadn't installed it. But Github has many versions available. Which one?

Doubletop12, 2023-08-26:
@Padlex Another look in Github and realise you have a copy of svg-to-gcode available I have done the pip install svg-to gcode and I got

```
Installing collected packages: svg-to-gcode
Successfully installed svg-to-gcode-1.5.4
```
But still nothing. I have no idea whet svg-to-gcode ended up but laser.py is in " C:\Users\Pete\AppData\Roaming\inkscape\extensions\laser"

So I copied the "laser" folder to C:\Program "Files\Inkscape\share\inkscape\extensions\laser" and now I get the J-Tech popup window. But clicking "Apply" gives me

[image]

Doubletop12, 2023-08-26:
Amongst other things I am getting this in the extension-errors.log when Inkscape opens up

Extension "J Tech Photonics Laser Tool" failed to load because a dependency was not met.
Dependency:
	type: file
	location: extensions
	string: inkex.py

The main extension folder for inkex exists here "C:\Program Files\Inkscape\share\inkscape\extensions" but there is no inkex.py file only inx.py

Doubletop12, 2023-08-26:
Because of this message in "C:\Users\Pete\AppData\Roaming\inkscape\extension-errors.log
Extension "J Tech Photonics Laser Tool" failed to load because a dependency was not met.
Dependency:
type: file
location: extensions
string: inkex.py

I put a dummy **_inkex.py_** into "C:\Users\Pete\AppData\Roaming\inkscape\extensions" and the message went away so is **_laser.py_** looking in the wrong place? 

I have nothing in my Windows paths that points to any of this stuff so should I have and what should it be?

Doubletop12, 2023-08-26:
This only works with Inkscape V1.1.2 (not 1.0 or 1.2 or 1.3)

closed 2023-08-26 by Doubletop12