JTechPhotonics/J-Tech-Photonics-Laser-Tool #92: Not working in Linux with Inkscape 1.3

RWL-69, 2023-12-06:
I have seen that the latest version of the plugin won't work with 1.3.x , this may be helpful.   I am testing with 1.3.2.

After trying to get this working, I have done all I can.  It still will not work.

Started with an error about missing inkex.py.  It now doesn't install with the default Inkscape install but with inksmoto.  The path for inkex.py is not part of the moto tools.  I still had to make a sim-link to the same directory as laser.py to get this far.

After getting to a point that I do get a working menu and I get the tools pop-up for setting the gcode for laser on and off, I get an error box.
I don't have any messages in the extension-errors.log

In the error pop-up box I get:
`
/home/robin/.config/inkscape/extensions/laser.py:1063: SyntaxWarning: invalid escape sequence '\s'
  if type_.lower() in re.split("[\s\n,\.]+", errors.lower()):
/home/robin/.config/inkscape/extensions/laser.py:1067: SyntaxWarning: invalid escape sequence '\s'
  elif type_.lower() in re.split("[\s\n,\.]+", warnings.lower()):
/home/robin/.config/inkscape/extensions/laser.py:1071: SyntaxWarning: invalid escape sequence '\s'
  elif type_.lower() in re.split("[\s\n,\.]+", notes.lower()):
/home/robin/.config/inkscape/extensions/laser.py:1193: SyntaxWarning: invalid escape sequence '\s'
  r = re.match("^\s*.\s*(\S+)", path.get("d"))
Traceback (most recent call last):
  File "/home/robin/.config/inkscape/extensions/laser.py", line 29, in <module>
    import simpletransform
  File "/usr/share/inkscape/extensions/inkex/deprecated-simple/simpletransform.py", line 11, in <module>
    from inkex.deprecated import deprecate
ModuleNotFoundError: No module named 'inkex.deprecated'; 'inkex' is not a package

`

I don't have any messages in the extension-errors.log

Looking through the install, I see that there is a directory /usr/share/inkscape/extensions/inkex

moseltronics, 2024-02-20:
Hi,
i didn't have the some problem, but had many others.
One reason was that i installed and reinstalled the old and the new version several times (Linux) and didn't realize that the extensions path changed each time. I don't know what is responsible for this.

So each time after a (re-)installation one should check the extensions path in the settings menu. (User extensions worked for me.)

RWL-69, 2024-03-12:
I found the issue I was having after multiple attempts today.  The web page(s) I was following didn't give me a current plugin.

[https://jtechphotonics.com/?page_id=1980](url)
[https://jtechphotonics.com/?page_id=2012](url)

I just did search and came up with them again.  

I downloaded the github version and it worked first try.  Much different menu than on the links I found.

[https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool](url)

So, the issue is bad references to the proper link.

closed 2024-03-12 by RWL-69